RobotLabLTH/SkiROS2 #35: Scene generation

alibinjunaid, 2020-10-12:
Hi,

i am trying to create a scene for the robot operation environment. I have defined the owl in Protege and then i launch it but only robot is added on the scene with skills list which i can see in WorldModel tab of Rviz. I have manually converted individuals to elements for accessing their properties. How can i use the owl to automatically add individuals to the WorldModel scene upon launching?

Thank you.

frovida, 2020-10-13:
Hi Ali, to load a scene you have to specify the "init_scene" param when you launch the world model (See tutorial 2, launch the system). After boot, you can also load/save a scene from the WorldModel tab of the gui. Note that the scene files must be in the directory specified in the world model "workspace_dir" parameter, and it should be in .turtle format.

alibinjunaid, 2020-10-13:
Dear Frovida,

Thank you for your comment. Indeed, we are specifying the "init_scene" param. However, could you please guide how to generate the scene file (in .turtle format) from Protege? or how it is being done?

Thank you.

alibinjunaid, 2020-10-28:
Thank you Frovida for your assistance and guidance.

closed 2020-10-28 by alibinjunaid