RobotLabLTH/SkiROS2 #44: Task planner unsupported operand type(s) for +: 'NoneType' and 'string'

pleconte87, 2021-01-27:
When I try to launch the task planner example with
`roslaunch skiros2_examples planning_example.launch` and I run "task_plan"
skill from skiros GUI with a Goal in PDDL format.
`skiros:contain skiros:LargeBox-80 skiros:Starter-145`

I get the following error message 
`Task planner unsupported operand type(s) for +: 'NoneType' and 'string' `
in the skiros GUI.

Any idea why this happens?  I installed the Fast Downward planner using the script in `skiros2/skiros2/scripts` before running.

Kind regards

pleconte87, 2021-01-27:
The first issue is solved by changing line 58 in task_planner.py
However now this is the log that I get:

```
10:32:40;0.0002;;-1;task_4;4;Running;-2;
10:32:40;0.1110;task_4;4;task_plan;5;Running;1;Planning...
10:32:40;0.1327;;-1;task_4;4;Running;-2;
10:32:40;0.1150;task_4;4;task_plan;5;Running;2;Planning...
10:32:40;0.1720;;-1;task_4;4;Running;-2;
10:32:40;0.1538;task_4;4;task_plan;5;Running;3;Planning...
10:32:40;0.2119;;-1;task_4;4;Running;-2;
10:32:40;0.1935;task_4;4;task_plan;5;Running;4;Planning...
10:32:40;0.2518;;-1;task_4;4;Running;-2;
10:32:40;0.2334;task_4;4;task_plan;5;Running;5;Planning...
10:32:40;0.2919;;-1;task_4;4;Running;-2;
10:32:40;0.2736;task_4;4;task_plan;5;Running;6;Planning...
10:32:40;0.3318;;-1;task_4;4;Running;-2;
10:32:40;0.3134;task_4;4;task_plan;5;Running;7;Planning...
10:32:40;0.3719;;-1;task_4;4;Running;-2;
10:32:40;0.3536;task_4;4;task_plan;5;Running;8;Planning...
10:32:40;0.4120;;-1;task_4;4;Running;-2;
10:32:40;0.3935;task_4;4;task_plan;5;Running;9;Planning...
10:32:40;0.4518;;-1;task_4;4;Running;-2;
10:32:40;0.4334;task_4;4;task_plan;5;Running;10;Planning...
10:32:40;0.4918;;-1;task_4;4;Running;-2;
10:32:40;0.4736;task_4;4;task_plan;5;Running;11;Planning...
10:32:40;0.5318;;-1;task_4;4;Running;-2;
10:32:40;0.5133;task_4;4;task_plan;5;Running;12;Planning...
10:32:40;0.5718;;-1;task_4;4;Running;-2;
10:32:40;0.5534;task_4;4;task_plan;5;Running;13;Planning...
10:32:40;0.6118;;-1;task_4;4;Running;-2;
10:32:40;0.5936;task_4;4;task_plan;5;Running;14;Planning...
10:32:40;0.6560;;-1;task_4;4;Failure;-1;End
10:32:40;0.6374;task_4;4;task_plan;5;Failure;-405;Error on execution: Traceback (most recent call last):
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/visitors.py", line 159, in postProcessNode
    state = self.postExecute(procedure)
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/skill_utils.py", line 294, in postExecute
    state = self._postExecute(skill)
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/skill_utils.py", line 285, in _postExecute
    return skill.tick()
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/skill.py", line 309, in tick
    res = self.execute()
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/skill.py", line 517, in execute
    res = self._instance.tick()
  File "/home/ros/catkin_ws/src/skiros2/skiros2_skill/src/skiros2_skill/core/skill.py", line 309, in tick
    res = self.execute()
  File "/home/ros/catkin_ws/src/skiros2_std_lib/skiros2_std_skills/src/skiros2_std_skills/task_planner.py", line 58, in execute
    return self.fail(self._action_msg, str(-101))
  File "/home/ros/catkin_ws/src/skiros2/skiros2_common/src/skiros2_common/core/abstract_skill.py", line 523, in fail
    assert type(code) == int
AssertionError
. Blackboard data: Goal:['(skiros:contain skiros:LargeBox-80 skiros:Starter-145)'] Robot: cora:Robot-81-:test_robot 
```

frvd, 2021-01-29:
@pleconte87  I tried and managed to reproduce your first error, but It disappeared while I was looking for it, and now I am not able to reproduce that again. 

First of all, remove the change you made to line 58, because is generating your second error.

Can you then set the task manager in verbose mode, by adding a parameter in the planning_example.launch like this:
```
<launch>
  <arg name="deploy" default="false" />
  <include file="$(find skiros2_examples)/launch/main.launch">
    <arg name="init_scene" value="planning_scene.turtle"/>
    <arg name="deploy" value="$(arg deploy)"/>
    <arg name="skill_list" value="[drive_fake, pick_fake, place_fake, task_plan]" />
    <arg name="gui" default="true"/>
  </include>
  <include file="$(find skiros2)/launch/task_mgr.launch">
      <arg name="deploy" value="$(arg deploy)"/>
      <arg name="verbose" value="true"/>
  </include>
</launch>
```
And then paste here the output on the task_manager terminal? Thanks!

pleconte87, 2021-02-01:
Hi, so the origin of the error was found.  The ./skiros/planner directory was not executable and the path in the .bashr was not correct.
By changing this, I could make it work.  Having maybe a clearer error for this can be interesting?

closed 2021-02-19 by pleconte87