RobotLabLTH/SkiROS2 #46: Multiple robots = multiple skill managers?

robberthofmanfm, 2021-03-16:
Hi there,

I have a setup with multiple robots, and want to separate the skills for each of them.

On the wiki, I read sentences wherein the skill manager node is referenced in singular, and in the [architecture](https://raw.githubusercontent.com/wiki/RVMI/skiros2/imgs/skiros2_architecture.png) there seems to be only one skill manager node.
> At deployment time the skill manager node loads the desired skills [...].

However, when building the launch file for my application, I encounter an argument for the skill manager node, [`robot_name`](https://github.com/RVMI/skiros2_template_lib/blob/6d803758619d097a23dd5e54984e9f5e55f0e35c/launch/main.launch#L22), singular. 
`      <arg name="robot_name" value="$(arg robot_name)"/>`
This makes me believe each robot should have its own skill manager node.

When I make two of these [`<include>` blocks](https://github.com/RVMI/skiros2_template_lib/blob/6d803758619d097a23dd5e54984e9f5e55f0e35c/launch/main.launch#L16-L23) in the launch file, with a different `libraries_list`, `skill_list` and `robot_name`, two skill manager nodes are indeed started and two separate xterms pop up, all skills are loaded fine, the rostopics come up for each of them and everything seems in perfect harmony.

However, as soon as I execute some of those skills from the GUI, it fails (`No plugin with name ... found`), because all skills execute on the skill manager node that loaded first, the second skill manager never does anything. 

How should I do this? Should I make two skill manager nodes, or is there a way to tell the skill manager node that I have two separate robots? What should I put for `robot_name` then?

Best regards,
Robbert

frvd, 2021-03-31:
Hi @robberthofmanfm ,

all your considerations are correct. The fact is, that the GUI can't currently handle multiple skill managers. If you want to use skiros with multiple skill managers, you have to create your own GUI or or modify the existing one.

matthias-mayr, 2022-12-02:
This issue seems answered and is stale. I am closing it. 
Feel free to re-open it if there are further questions.

closed 2022-12-02 by matthias-mayr