RobotLabLTH/SkiROS2 #51: Understanding world_model_interface._resolve_elements2

robberthofmanfm, 2022-04-11:
Hi,

Last Friday, we were getting a cryptic error resulting from [an argumentless "raise"](https://github.com/RVMI/skiros2/blob/master/skiros2_world_model/src/skiros2_world_model/ros/world_model_interface.py#L603)  in world_model_interface. In python2, that results in ```TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType```.

This was resolved by simply increasing [`iters = 5`](https://github.com/RVMI/skiros2/blob/master/skiros2_world_model/src/skiros2_world_model/ros/world_model_interface.py#L599) to 'iters = 10'. The problem is: we have no clue if 10 is a sensible choice or whether it will scale to large world models, because we don't understand what `_resolve_elements2` is doing.

The brief description for this function is to "Find all possible inputs for one or more keys of a skill parameter handler".
Does that mean that when you have a certain skill where parameters need to be inferred, this method will go into the world model to find out which world elements can be used to fill in those parameters?

Kind regards,
Robbert

frvd, 2022-06-23:
Hi @robberthofmanfm , 

sorry for coming back so late, I completely missed this message.

But yes, the scope of the function is exactly what you pointed out : finding how to fill the parameters that are left blank, but are mandatory for the skill execution. It uses the element type AND the pre-conditions as filters, or in other words , it will only select parameters that fulfil the pre-conditions. 
The conditions make pairs of interdependent parameters, but then you have to check the crossconnections , making sure you select inputs that fulfil all pre-conditions. This is what the function is doing. If you have a lot of preconditions, this naturally would require more iterations. I think the iterations limiter could be removed, looking at it now it doesn't make sense it is there.

closed 2022-06-24 by robberthofmanfm

matthias-mayr, 2022-06-24:
> I think the iterations limiter could be removed, looking at it now it doesn't make sense it is there.

What would happen if it's removed and some configuration can not be resolved?

Maybe a reasonable middle ground would be to set it to a number that is high enough so it should never be hit by any user.

frvd, 2022-06-27:
Sure that would do too. But then we should definitely raise a meaningful error