RobotLabLTH/SkiROS2 #52: Problem with inferred parameters

toondaemenFM, 2022-10-31:
Hi,

I experience the following problem when I want to use parameters of the type _inferred_:

Consider two skills that implement the functionality to respectively pick a part from a container, and place a part in a container. Both skills have a very similar SkillDescription:

```
class PickSkill(SkillDescription):
    def createDescription(self):
        #=======Params=========
        self.addParam("Container", Element("skiros:Container") , ParamTypes.Required)
        self.addParam("Part", Element("skiros:Part") , ParamTypes.Inferred)

        #=======PreConditions=========
        self.addPreCondition(self.getPropCond("PartPresent", "hasObject", "Part", "=", True, True))
        self.addPreCondition(self.getRelationCond("InferPart", "skiros:contain", "Container", "Part", True))

        #=======PostConditions=========
        self.addPostCondition(self.getPropCond("PartPresent", "hasObject", "Part", "=", False, True))
```

```
class PlaceSkill(SkillDescription):
    def createDescription(self):
        #=======Params=========
        self.addParam("Container", Element("skiros:Container") , ParamTypes.Required)
        self.addParam("Part", Element("skiros:Part") , ParamTypes.Inferred)

        #=======PreConditions=========
        self.addPreCondition(self.getPropCond("PartPresent", "hasObject", "Part", "=", False, True))
        self.addPreCondition(self.getRelationCond("InferPart", "skiros:contain", "Container", "Part", True))

        #=======PostConditions=========
        self.addPostCondition(self.getPropCond("PartPresent", "hasObject", "Part", "=", True, True))
```
The skills expect the container where we want to pick or place a part to be specified, and will infer the Part instance from the world model through the pre-condition labeled "InferPart": It will look which Part instance is related to the given Container instance through the predicate 'skiros:contain'. The Part instance here is actually a placeholder for a part, and the actual presence of a part is indicated by a parameter 'hasObject', that is being checked by the condition labeled "PartPresent". 

This works as expected, If I run the pick skill or place skill, the right Part instance is inferred, and the skill only executes when the condition labeled 'PartPresent' is fulfilled. 

however, it fails when I make a pick-and-place skill, which sequences the pick skill and place skill respectively:
```
skill(
         self.skill('PickSkill', 'pick_skill', specify={Container: container_pick}),
         self.skill('PlaceSkill', 'place_skill', specify={Container: container_place})
)
```

What happens now is that SkiROS fails to infer the part that is related to the container where we want to **place** the part, because it tries to find a relation between the container we specify for placing, and the part that was previously inferred in the **picking** skill. I suspect that it somehow is caused by world model elements that are listed on the **Blackboard**, because:
- If I print the contents of the blackboard between execution of the pick skill and the place skill, I see the elements that I inferred in the pick skill.
- If I choose an unique name for all parameters of the type _inferred_, across all SkillDescriptions (i.e. explicitly using _PartPick_ and _PartPlace_ instead of reusing the label _Part_), Inference is done like expected.

When I add parameters, I tried to specify parameter options, like _Consume_ or _Unspecify_. According to the [source code](https://github.com/RVMI/skiros2/blob/893bd37c7b83dd31c7b0bef48f950c0c69f7e526/skiros2_common/src/skiros2_common/core/abstract_skill.py#L17) these options should result in the world model elements being removed from the blackboard after skill execution.

However, I notice that it does not work when two skills are sequenced...

Could you please provide me some information on how I should address my issue?
I also would be happy if you could provide me some good examples on inference and resolving from the world model, as these are really powerful features.

Regards,

Toon Daemen
Robotics Researcher at Flanders Make, Belgium

matthias-mayr, 2022-12-02:
Excuse the delay. We were facing a similar issue and went the first way to use different parameter names. Your assessment of the situation is correct. SkiROS2 has a global blackboard right now and names can collide.

The global blackboard can also be used as a feature to share data, e.g. between a camera a processing skill, but it of course also has downsides.

We haven't used the parameter options yet and I'm having some issue to quickly see where they would be applied. I am spending some more time with the source code soon, so then I might be able to tell a bit more.

frvd, 2023-03-07:
Hi @toondaemenFM,
in this case the right solution is to remap the place parameter to another key. In general, it is always better (it is more flexible) to work with remaps rather then specifying the parameters with specify, because specify is used only when expanding the skill, while remaps will work with parameters in the blackboard that can be updated during execution.

frvd, 2023-03-07:
Btw, from what I recall the Options are ignored by the default Executor visitor. if you want to use them you would need to implement it in the visitor. Maybe you can cherry pick the implementation from one of the other visitors, but I am not sure