RobotLabLTH/SkiROS2 #57: Re-evaluation of inferred parameters when looping over skills

toondaemenFM, 2023-02-16:
Hi,

I created a skill that makes a robot pick parts from a container. In the world model, part instances are associated to a container instance through the following relations:

|   Scene-0	|hasA   	|  Container-175 	|
|---	|---	|---	|
| Container-175  	| contain   	| Part-184  	|
| Container-175 	| contain 	        | Part-186   	|
| Container-175  	| contain  	| Part-188  	|
| Container-175  	| contain  	| ...  	|

I created a skill description where I specify a robot and a container, and I let SkiROS automatically parametrize the Part parameter by imposing a precondition. This way, the skill chooses one of the parts in the container to pick:

```
class PickPart(SkillDescription):
        def CreateDescription(self):
                 #=======Params==
                 self.addParam('Robot', Element('cora:Robot'), ParamTypes.Required)
                 self.addParam('Container', Element('skiros:Container'), ParamTypes.Required)
                 self.addParam('Part', Element('skiros:Part'), ParamTypes.Inferred)

                 #=======PreConditions==
                 self.addPreCondition(self.getRelationCond('InferPart', 'skiros:contain', 'Container', 'Part', True))
```

During skill execution, the relation of the Part with the Container is broken once the robot picks the part from the container. This way, each time we launch the skill, another part instance is found when the parameter Part is autoparametrized through the preconditions. By launching the skill several times, the robot unloads the container.

However, we want the iterative behaviour to be automatic, i.e. instead of manually launching the picking skill several times in a row, it should automatically start looping the skill until there are no more Part instances linked to the Container. 
Let's try to follow my way of reasoning:

- I tried to include some kind of looping in the ```skill()``` statement in the skill's ```expand(self, skill)``` method, but that didn't work out. 
- I thought it might be a good solution to create a dedicated skill processor here, so I created a new processor ```Repeating()``` which is basically the same as ```SerialStar()```, but it restarts the sequence of children if they all succeeded.
- When SkiROS fails to automatically parametrize a parameter, it returns a status called ```Status.Idle```. So, since I want to loop over the skill as long as the preconditions can render a suitable parametrization, I implemented the ```Repeating()``` processor such that it exits once the skill I want to loop returns ```Status.Idle```. This way, I have a processor that keeps on executing the same skill as long as a parametrization can be obtained.

In short, I have now something like this:
```
class unload_container(SkillBase): 
        def expand(self, skill):
                skill.setProcessor(Repeating())
                skill(
                       self.skill('PickPart', 'pick_part', specify={'Robot': robot, 'Container': container})
                 )
```

The skill PickPart is now executed, and SkiROS autoparametrized a suitable Part instance that has the desired relation to the Container. Also, after successful execution, the  ```Repeating()``` processor tries to restart the skill PickPart again.
However, The skill returns ```Status.Idle``` the second time, and I receive the message that the relation for the given Part instance is not met. Here, the message mentions the Part instance that has been autoparametrized in the previous iteration. 
Actually, I want that the skill re-evaluates its pre-conditions each time the skill is triggered, and that the parameters are updated accordingly.

Since the first iteration always involves a suitable autoparametrization, and since I can entirely unload the container by manually pressing the playbutton several times, I definitely think it should be possible somehow, although I don't manage to find the right way to do it. 
Thus far, I have been attracted to 
- [```def process(self, procedure):```](https://github.com/RVMI/skiros2/blob/de00abafffe78c8da6ba35e8259d6ad075ec72b6/skiros2_skill/src/skiros2_skill/core/visitors.py#L44) in visitors.py
- [```def specifyParams(self, input_params, keep_default=True):```](https://github.com/RVMI/skiros2/blob/de00abafffe78c8da6ba35e8259d6ad075ec72b6/skiros2_common/src/skiros2_common/core/abstract_skill.py#L480) in abstract_skill.py, skill.py and params.py.
- [```def checkRules(self, p1, p2):```](https://github.com/RVMI/skiros2/blob/de00abafffe78c8da6ba35e8259d6ad075ec72b6/skiros2_skill/src/skiros2_skill/core/visitors_optimizer.py#L58) in visitors_optimizer.py

Could you please point me in the right direction?

Regards,
Toon Daemen
Robotics engineer at Flanders Make, Belgium

matthias-mayr, 2023-02-22:
Great description Toon.
I think I understood the problem, but we're very busy with some papers in the next few weeks unfortunately.

An easy workaround that I could think about is to make a skill primitive that would resolve "Part" and set it on the blackboard instead of using the autoparameterization of the parent skill. That primitive could be the first one in a pick sequence and would fail if there is no part and succeed if there is still a part.
It is not exactly the same, but should be a quick solution to achieve the same thing.

frvd, 2023-03-07:
Another equivalent solution could be to add a skill to forget about the part so at next iteration the autoparametrize would kick in again.

matthias-mayr, 2023-05-08:
I spent some time with this since I have a similar scenario today. I tried to avoid "my" solution by writing a primitive and looked into a more sound solution that uses the autoparameterization capability.

I followed @frvd's suggestion to unset the blackboard parameter. The first thing I learned is that the [`ParamOptions`](https://github.com/RVMI/skiros2/blob/ae3f0400bc2e6feaf34c9ad560779857c36eb7e0/skiros2_common/src/skiros2_common/core/abstract_skill.py#L22) which for example offer this
`Unspecify: (postCondition) The parameter is unspecified (removed from the Blackboard) after execution`
only create a post-condition, but not actually implement that behavior. Nevertheless I made #62 on the way.

Then I basically implemented an "unset BB parameter" primitive like this:
```python
class bb_unset_param(PrimitiveBase):
    def createDescription(self):
        self.setDescription(BbUnsetParam(), self.__class__.__name__)    

    def execute(self):
        # Way 1:
        # self.params["Parameter"].unset()
        # Way 2:
        param = self.params["Parameter"].value
        param._id = ""
        self.params["Parameter"].value = param
        return self.success("Unset succeeded")
```

**Way 1: Unset:**
First I tried to unset the parameter. This means that the parameter does not have any values anymore.

First this looks fine. In the postprocessing step of a node tick, the parameters of that node [are merged](https://github.com/RVMI/skiros2/blob/ae3f0400bc2e6feaf34c9ad560779857c36eb7e0/skiros2_skill/src/skiros2_skill/core/skill_utils.py#L297).
This means that our parent comes with `ParamA, ParamB, ParamC` and if we "unset" `ParamB`, the parent would only have `ParamA and ParamC` remaining.

However if we follow our tick-postprocessing one level higher up, our parent node would then claim that only `ParamA and ParamC` exist anymore in its scope and our "unset" would not progress any further.

My attempt to tackle this is to [remove the lines that would remove our unset `ParamB`](https://github.com/RVMI/skiros2/blob/ae3f0400bc2e6feaf34c9ad560779857c36eb7e0/skiros2_common/src/skiros2_common/core/params.py#L230-L232). This way the non-existing value of our blackboard parameter is propagated through all merges. 

However, it turns out that various parts of the code base rely on parameters having their values and while I added some additional checks it felt difficult to oversee the implications.

**Way 2: Removing the ID**:
Eventually I also checked how parameters for the autoparameterization are identified and one key difference is that they do not have an ID. So I set the ID to an empty string.

At first this does not seem to work, because a WM update is performed right after the tick. If the parameter does not have any ID, a new WM element is created and we have not succeeded to trigger autoparameterization. However, #63 prevents that this happens.

With the change in #63  and the usage of the primitive above, I am able to trigger the desired behavior of unsetting some blackboard parameter and trigger the autoparameterization.

----

While it works it feels a bit hacky. At this point I am not entirely sure about the implications.

Also in general, it feels like it would be a more accessible programming paradigm if the `ParamOptions` would directly implement their behaviors. But that feels a bit more like being on a wishlist right now.

matthias-mayr, 2023-06-23:
Way2 can also cause troubles. When autoparameterizing, this BB parameter would be used again & the ID is used as an indicator whether it was set.
Example: At one point we would check a relation such as
```
Object-1 skiros:hasA ApproachPose-1
```
Now we set the ID of "ApproachPose" to nothing & there is code that treats this as a wildcard. We can end up in situation in which we want to check whether there is a relation between "Object-1" and our "ApproachPose" for which we just deleted the ID.
Since our "ApproachPose" without ID is treated as wildcard, this means that if "Object-1" has any relations with the right predicate, this is interpreted as it having a relation to our "ApproachPose" without ID. I am sure the behavior can be changed where this happens, but some other approach would probably make more sense.

matthias-mayr, 2023-06-26:
I looked into a better solution today and it's as easy as setting a blackboard parameter to an empty element of the correct type.
So an unset primitive can look like this:
```python
class BbUnsetParam(SkillDescription):
    """
    @brief Removes a parameter from the blackboard, e.g. to trigger the autoparameterization
    """
    def createDescription(self):
        # =======Params=========
        self.addParam("Parameter", Element("sumo:Object"), ParamTypes.Required)

class bb_unset_param(PrimitiveBase):
    def createDescription(self):
        self.setDescription(BbUnsetParam(), self.__class__.__name__)    

    def execute(self):
        param = self.params["Parameter"].value
        new_el = Element()
        new_el.type = param.type
        self.params["Parameter"].value = new_el
        return self.success("Unset of {} succeeded".format(param.label))
```
As said, this could also be integrated inside SkiROS itself, but this primitive works too. I am probably going to send this as a PR to the std lib.