RobotLabLTH/SkiROS2 #82: How can I deal with symmetric property in world model

HB-Liang, 2023-11-30:
Hi, I am using skiros2 for a simple navigation planning. I uses a connection relation that shows the connectivity between way-points and the relation should be symmetric. But in precondition check, it seems to be a directional relation.
E.g. I have a graph showing
point1 connected point2
point2 connected point3
point3 connected point1,
and with the precondition " Connected, ns:hasConnection, Start, Target, True", and point1 as Start, point3 as Target, I can get a plan that goes from point1 to point2 and point3. However, a plan that goes from point1 to point3 is also possible.
How can I check the symmetric property in condition check? Sorry for non-native expressions.
Thanks.

matthias-mayr, 2023-12-02:
You are correct that the order in relations needs to be right. This often makes sense, because `ObjA onTop ObjB` is not the same as `ObjB onTop ObjA`.

There are several ways, your use-case could be tackled:
1. **Complete the world model with relations.** This is the easiest thing to do. You can write a simple python script that checks all the connections and sets the reverse connections if they are not there yet. A big plus on this is that no other modifications are needed
2. **Improved pre-condition check**: There is a special condition type that can connect two conditions. The [ConditionOr](https://github.com/RVMI/skiros2/blob/b183864744be536c800bd1b00021dde0122dc9d4/skiros2_common/src/skiros2_common/core/conditions.py#L84-L158) that can connect several conditions. Its usage is shown [here](https://github.com/RVMI/skiros2/blob/b183864744be536c800bd1b00021dde0122dc9d4/skiros2_common/src/test/test_conditions.py#L21C1-L31). It could be used to check either relation.
3. **Extend the relation-condition check**: The relation condition check could also be extended with an additional argument to check in either direction.

I am not sure how well options 2-3 go with the planner. I haven't seen anyone doing 2. and at least 3. would require coding. My recommendation would be to just complement the relations. It's the safest and easiest.

HB-Liang, 2023-12-04:
Thanks for your recommendations! This offers some  solutions to me. I assume that the Option 1 would works well and treat it as the last choice because in my point of view, it would almost double the world model with completed relations.
I tried Option 2 in my case, using ConditionOr to combine both direction with addCondition.
```
self.connections = ConditionOr(True)
self.Target2Start = ConditionRelation("Connected", "ns1:hasConnection", "TargetLocation", "StartLocation", True)
self.Start2Target = ConditionRelation("Connected", "ns1:hasConnection", "StartLocation", "TargetLocation", True)
self.connections.addCondition(self.Target2Start)
self.connections.addCondition(self.Start2Target)
```
So far, the code seems to work well, `self.connections` combine both direction with "or" and I get the description as expect in the terminal:

> [or] ( [Connected) TargetLocation-ns1:hasConnection-StartLocation(True) [Connected] StartLocation-ns1:hasConnection-TargetLocation(True) )

I uses `self.addPreCondition(self.connections)` and expects a both direction relation check. In `self._pre_condition`, `self.connections` is included.
However, when a goal was given, the generated domain file did not include the condition in `self.connections`. Also, when I ran a single skill, the error "can't autoparametrize param" occured. Would you please give more details?
Regards.

matthias-mayr, 2023-12-06:
> Thanks for your recommendations! This offers some solutions to me. I assume that the Option 1 would works well and treat it as the last choice because in my point of view, it would almost double the world model with completed relations.

I would not really worry about the extra relations, but I can see why you want to check out the other options first.

> However, when a goal was given, the generated domain file did not include the condition in self.connections

That could be. I haven't taken a look at the code that translates preconditions to PDDL

> what's more, all the connection relations are not generated in problem file. 

That could be an effect of the missing preconditions. iirc only things that play a role end up in the problem. So if the connection is not relevant, because the nested precondition is not processed, the connection would not be in the problem file

> Also, when I ran a single skill, the error "can't autoparametrize param" occured. Would you please give more details?

This typically means that some inferred connection could not be resolved. To really help you with that, it would be helpful to see the skill description and the exact error message.