RobotLabLTH/SkiROS2 #84: Installing td task planner

TimKnissel, 2024-01-02:
It seems like the uni-freiburg has changed their download address for the tfd planner.

In the "install_fd_task_planner.sh" I had to change the url in line 23

from: "https://gki.informatik.uni-freiburg.de/tools/tfd/downloads/version-0.4/${tfd}.tgz"
to: "https://tfd.informatik.uni-freiburg.de/downloads/version-0.4/${tfd}.tgz"

matthias-mayr, 2024-01-04:
Thanks for the input.

Unless they also updated also the code, installation of the planner would need to change to a version that supports Python3. #78 implements a switch to an adapted version, but I want to run the installation procedure with the PR once again before merging it.

matthias-mayr, 2024-01-19:
#78 is merged now and looks like it's working. If there are any problem, this can still be reopened.

closed 2024-01-19 by matthias-mayr