RobotLabLTH/SkiROS2 #92: Python dependencies "warpt" and "inflection" in 22.04 ROS 2

matthias-mayr, 2024-03-01:
In ROS 1 we installed `wrapt` and `inflection` as pip packages with `requirements.txt`.   
In ROS 2 Humble on 22.04 we used the system packages so far. This should be a more stable choice, however, there is currently no procedure to install them.

The easiest way is to have again an install script and/or some instructions.   
The nice way would be to make a PR for `rosdep` and [make a PR for a new rule](http://docs.ros.org/en/independent/api/rosdep/html/contributing_rules.html) and put them in the `package.xml`

zanxueyan, 2024-08-12:
Currently, does skiros2 support ROS 2 Humble? I look at the code is still using ros1 library ah, where to download support ROS 2 Humble project? Ask for advice

bchen32, 2024-09-10:
Seems like in the latest commit to the skiros2 master branch somebody added wrapt and inflection to the package.xml, but I don't think a rosdep rule has been made for them yet so installation (`rosdep install --from-paths . --ignore-src --rosdistro=$ROS_DISTRO -y`) is failing for me right now

matthias-mayr, 2024-09-10:
In ROS 1 or ROS 2? In humble as well as Jazzy, I can resolve `python3-wrapt` and `python3-inflection`

bchen32, 2024-09-10:
> In ROS 1 or ROS 2? In humble as well as Jazzy, I can resolve `python3-wrapt` and `python3-inflection`

Might be a ROS1 issue then? I'm on noetic, ubuntu 20.04

marcusklang, 2024-09-11:
> > In ROS 1 or ROS 2? In humble as well as Jazzy, I can resolve `python3-wrapt` and `python3-inflection`
> 
> Might be a ROS1 issue then? I'm on noetic, ubuntu 20.04

I just tested it in a fresh docker container derived from the `ros:noetic` image.

This container with a clone of this repo resolved `python3-wrapt` and `python3-inflection` when running a similar command to yours:
`rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y` although this was meant to be run from the workspace root.

This container reports running Ubuntu 20.04.6 LTS inside.

bchen32, 2024-09-11:
Sorry, I realized I simply forgot to do rosdep update.

closed 2025-06-09 by marcusklang