RobotLabLTH/SkiROS2 #106: ros2 launch error

pdl0x0lbq, 2024-03-19:
I encountered an error when using the command
 `ros2 launch skiros2 skiros2 launch. py`
`[INFO] [skill_manager_node-4]: process started with pid [76913]
[static_transform_publisher-1] [WARN] [1710860048.171697259] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-1] [INFO] [1710860048.177983790] [static_transform_publisher_SzrOYGeJA5UMXorC]: Spinning until stopped - publishing transform
[static_transform_publisher-1] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-1] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-1] from 'map' to 'world'
[skill_manager_node-4] [INFO] [1710860048.430873862] [skiros]: [WorldModelInterface] Waiting wm communications...
[world_model_server_node-2] [INFO] [1710860048.457418128] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/aqr.owl
[world_model_server_node-2] [INFO] [1710860048.460694688] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.460835817] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/skirosExtra.owl
[world_model_server_node-2] [INFO] [1710860048.464148045] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.464282088] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/skiros.owl
[world_model_server_node-2] [INFO] [1710860048.476922042] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.477104941] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/cora.owl
[world_model_server_node-2] [INFO] [1710860048.481631385] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.481773360] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/sumo.owl
[world_model_server_node-2] [INFO] [1710860048.486210165] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.486350717] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/coraX.owl
[world_model_server_node-2] [INFO] [1710860048.489148390] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.489286973] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/RParts.owl
[world_model_server_node-2] [INFO] [1710860048.492493850] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.492672414] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/aqr.owl
[world_model_server_node-2] [INFO] [1710860048.494596581] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.494735064] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/skirosExtra.owl
[world_model_server_node-2] [INFO] [1710860048.498634769] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.498774867] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/skiros.owl
[world_model_server_node-2] [INFO] [1710860048.510313238] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.510492986] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/cora.owl
[world_model_server_node-2] [INFO] [1710860048.514824144] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.514962779] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/sumo.owl
[world_model_server_node-2] [INFO] [1710860048.521305150] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.521865754] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/coraX.owl
[world_model_server_node-2] [INFO] [1710860048.526339614] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.526898154] [skiros]: [WorldModel]        Loading ontology: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl/IEEE-1872-2015/RParts.owl
[world_model_server_node-2] [INFO] [1710860048.531757566] [skiros]: [WorldModel]        Set id: aqr for ontology: http://www.semanticweb.org/sw/ontologies/2024/2/aqr#
[world_model_server_node-2] [INFO] [1710860048.532277161] [skiros]: [WorldModelServer]  Workspace folder: /home/sw/catkin_ws/install/skiros2/share/skiros2/owl
[world_model_server_node-2] [INFO] [1710860048.643700850] [skiros]: [load_reasoner] Loaded AauSpatialReasoner
[skill_manager_node-4] [INFO] [1710860048.681545902] [skiros]: [WorldModelInterface] Wm communications active.
[skill_manager_node-4] [INFO] [1710860048.684535398] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.685286100] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.688800559] [skiros]: Service call to wm/modify with spining
[world_model_server_node-2] [INFO] [1710860048.690026361] [skiros]: AauSpatialReasoner  warn: Adding relation to http://www.inf.ufrgs.br/phi-group/ontologies/cora.owl#Robot-1-skiros:test_robot
[world_model_server_node-2] [INFO] [1710860048.690350739] [skiros]: [AauSpatialReasoner] Publishing http://www.inf.ufrgs.br/phi-group/ontologies/cora.owl#Robot-1-skiros:test_robot parent: map
[skill_manager_node-4] [INFO] [1710860048.694659231] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.698367113] [skiros]: Service call to wm/modify with spining
[skill_manager_node-4] [INFO] [1710860048.709583514] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.713332090] [skiros]: Service call to wm/modify with spining
[skill_manager_node-4] [INFO] [1710860048.722714333] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.726762999] [skiros]: Service call to wm/modify with spining
[skill_manager_node-4] [INFO] [1710860048.734674630] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.738323728] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.741285463] [skiros]: Service call to wm/modify with spining
[world_model_server_node-2] [INFO] [1710860048.743336291] [skiros]: AauSpatialReasoner  warn: Adding relation to skiros:Location-5-skiros:unknown_location
[skill_manager_node-4] [INFO] [1710860048.748570207] [skiros]: Service call to wm/scene/set_relation with spining
[skill_manager_node-4] [INFO] [1710860048.749416604] [skiros]: Service call to wm/get with spining
[skill_manager_node-4] [INFO] [1710860048.750598859] [skiros]: [SkillManager]      Registered robot http://www.inf.ufrgs.br/phi-group/ontologies/cora.owl#Robot-1-skiros:test_robot
[skill_manager_node-4] [INFO] [1710860048.751036915] [skiros]: Service call to wm/modify with spining
[skill_manager_node-4] [INFO] [1710860048.755871856] [skiros]: [LoadLibrary]       
[skill_manager_node-4] Traceback (most recent call last):
[... cut ...]
[skill_manager_node-4]     return _bootstrap._gcd_import(name[level:], package, level)
[skill_manager_node-4]   File "<frozen importlib._bootstrap>", line 1047, in _gcd_import
[skill_manager_node-4]   File "<frozen importlib._bootstrap>", line 981, in _sanity_check
[skill_manager_node-4] ValueError: Empty module name
[ERROR] [skill_manager_node-4]: process has died [pid 76913, exit code 1, cmd '/home/sw/catkin_ws/install/skiros2_skill/lib/skiros2_skill/skill_manager_node --ros-args --params-file /tmp/launch_params_jy8ms9_4'].
`

matthias-mayr, 2024-03-19:
Thank you for reporting this.

The log is a bit difficult to read, in the future it would be great if it could be formatted [with triple backticks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code) as this tends to preserve the linebreaks.

At first sight it looks like it has been started with an empty skill list. And there should of course be a better way to communicate this to the user. The ROS 2 port is also not completely done and tested, but besides the issues that are reported here, it should work.

To verify, can you clone the [test lib](https://github.com/RVMI/skiros2_test_lib/) and launch the launch file from there? This would load some skills.

pdl0x0lbq, 2024-03-20:
> Thank you for reporting this.
>
> The log is a bit difficult to read, in the future it would be great if it could be formatted [with triple [...]

Thank you very much for patiently reading my terrible log. Skiros2utest_lib can be started, but an error appears to be related to the missing action message type.
```
[rqt_gui-3] [ERROR] [1710921502.514864704] [get_message_class]: Malformed msg message_type: skiros2_msgs/action/TestAction_FeedbackMessage
[rqt_gui-3] TopicInfo.__init__(): topic "/test_action/_action/feedback": can not get message class for type "skiros2_msgs/action/TestAction_FeedbackMessage"

```
I did not find TestAction_feedbackMessage in the skiros2/skiros2_msgs/action/folder

pdl0x0lbq, 2024-03-21:
> > I did not find TestAction_feedbackMessage in the skiros2/skiros2_msgs/action/folder我在 skiros2/skiros2_msgs/action/ 文件夹中没有找到 TestAction_feedbackMessage
> 
> This is weird. The `test.action` is there and it should be built. Some follow-up questions then:这很奇怪，这个项目就在那里，应该建起来。
> 
> 1. Was the workspace source after building it for the first time?第一次构建工作空间源代码后是否还存在？
> 2. Is the action listed in  中列出的操作是否正确？`ros2 action list`
> 3. If yes, what does `ros2 action info skiros2_msgs/TestAction` say?   如果有，是怎么说的？
The source code for the first workspace build exists.

Running `ros2 action list`will display `/test_action`

Running `ros2 action info skiros2_msgs/TestAction`will display 
```
Action: skiros2_msgs/TestAction
Action clients: 0
Action servers: 0
```

matthias-mayr, 2024-03-25:
Hm. I do not see a reason why it should not work.

Can you do the following steps:
1. Clean the colcon workspace
```
cd ~/catkin_ws
rm -r build
rm -r install
rm -r log
```
2. Build the colcon workspace. We use `colcon build --symlink-install`
   Check whether there are any errors in this step.
3. Source the built workspace `source ~/catkin_ws/install/setup.bash`
4. Run the test launch file again in the same terminal where steps 1-3 happened

matthias-mayr, 2024-08-16:
I assume that this is solved. If not and the same issue persists, feel free to open this issue again.

closed 2024-08-16 by matthias-mayr