RobotLabLTH/SkiROS2 #111: Planning Example Doesn't Work Consistently

bchen32, 2024-09-19:
I'm running the task planning example from the skiros2_examples repo and it randomly fails sometimes. I'm on ROS1 Noetic. I run `roslaunch skiros2_examples planning_example.launch`, then run the task_plan skill with this PDDL goal `(skiros:contain skiros:LargeBox-80 skiros:Starter-145)`. However, sometimes the skill will randomly fail on pick_fake or place_fake even though the primitives have succeeded.
[image]
I can't glean much information from the logs, but here is a snippet from the end of the logs. Interestingly, when I try to set verbose=true in the launch files, or use pdb to step through the code, I cannot reproduce this failure.

15:25:39;0.9690;pick_fake;7;wait;15;Running;26;Waiting 1.0
15:25:39;0.0000;pick_fake;7;wm_move_object;16;Idle;0;
15:25:39;0.0000;task_plan;5;drive_fake;8;Idle;0;
15:25:39;0.0000;task_plan;5;place_fake;9;Idle;0;
15:25:39;3.0942;;-1;task_4;4;Failure;-1;End
15:25:39;3.0798;task_4;4;task_plan;5;Failure;1;End
15:25:39;1.3673;task_plan;5;drive_fake;6;Success;1;End
15:25:39;1.0039;drive_fake;6;wait;10;Success;28;Done
15:25:39;0.0733;drive_fake;6;wm_set_relation;11;Success;1;Unset cora:Robot-53-skiros:at-skiros:Location-57
15:25:39;0.0915;drive_fake;6;wm_set_relation;12;Success;1;Set cora:Robot-53-skiros:at-skiros:LargeBox-51
15:25:39;1.1223;task_plan;5;pick_fake;7;Failure;1;End
15:25:39;1.0090;pick_fake;7;wait;15;Success;27;Done
15:25:39;0.0682;pick_fake;7;wm_move_object;16;Success;1;skiros:Starter-145 moved from skiros:LargeBox-51 to rparts:GripperEffector-56.
15:25:39;0.0000;task_plan;5;drive_fake;8;Idle;0;
15:25:39;0.0000;task_plan;5;place_fake;9;Idle;0;

matthias-mayr, 2024-09-24:
Thanks for this bug report. My guess is that a postcondition check is failing.

Does your skiros2_std_lib contain this commit? https://github.com/RVMI/skiros2_std_lib/commit/0c38b5e1b5339f745bc893888f7ff5691a3489d3  
If not, it would be great it you could pull and report back if it still happens for you.

bchen32, 2024-09-24:
@matthias-mayr Yes, I do have that commit. In a different project, postcondition checks that fail will have a log message like "error: ConditonCheck failed" but I don't see any such message here.