RobotLabLTH/SkiROS2 #117: ConditionProperty where desired_state is the value of another parameter?

bchen32, 2024-11-21:
If I have a primitive which takes in an int parameter `x` and an Element parameter `y`. Is there a way to create a post condition check that a property of `y` equals the value of `x`?

matthias-mayr, 2024-11-21:
Good question. Unfortunately this is currently not possible, but it would be a good thing to add.

The workaround that is definitely possible is to either incorporate this comparison inside the execute function of the existing primitive or to write a dedicated primitive skills that checks this.

bchen32, 2024-11-21:
What I ended up doing was just adding another condition class that was essentially the same as ConditionProperty but uses a param_name instead of value.

matthias-mayr, 2024-11-26:
Ah yes, that of course works as well. I was just thinking about out-of-the-box solutions.

Feel free to make it nice and do a PR. I'll happily take a look.